TixiaoShan/LVI-SAM #1: Can this work be used without IMU?

xiaobrnbrn, 2021-04-25:
Hello, I would like to know whether this work can be used for the fusion of only camera and lidar or worked with a 6-axis IMU?

TixiaoShan, 2021-04-25:
An IMU is a must. The camera module can work with a 6-axis IMU. The lidar module currently needs a 9-axis IMU, but it can support a 6-axis with some code changes.

xiaobrnbrn, 2021-04-25:
> An IMU is a must. The camera module can work with a 6-axis IMU. The lidar module currently needs a 9-axis IMU, but it can support a 6-axis with some code changes.

Thanks, I got it!

closed 2021-04-25 by xiaobrnbrn

GuoFeng-X, 2021-05-06:
> > An IMU is a must. The camera module can work with a 6-axis IMU. The lidar module currently needs a 9-axis IMU, but it can support a 6-axis with some code changes.
> 
> Thanks, I got it!

hello, please How to modify the code to support 6-axis IMU?

zuowei1998, 2022-01-03:
Hello, do you know how to modify he code to support 6-axis IMU? Thanks!