TixiaoShan/LVI-SAM #2: 使用handheld数据测试效果发生多次跳变

dzy130120, 2021-04-25:
使用handheld.bag实验效果发现偶尔会出现跳变，不知是否针对这个数据集程序中的参数还需要调整哪些？

slam-art, 2021-04-25:
I found a solution, you can try to reduce the playback speed of the bag file, refer to this [blog](https://blog.csdn.net/learning_tortosie/article/details/116051761?spm=1001.2014.3001.5501) for details.

slam-art, 2021-04-25:
> 使用handheld.bag实验效果发现偶尔会出现跳变，不知是否针对这个数据集程序中的参数还需要调整哪些？

I found a solution, you can try to reduce the playback speed of the bag file, refer to this [blog](https://blog.csdn.net/learning_tortosie/article/details/116051761?spm=1001.2014.3001.5501) for details.

dzy130120, 2021-04-25:
> > 使用handheld.bag实验效果发现偶尔会出现跳变，不知是否针对这个数据集程序中的参数还需要调整哪些？
> 
> I found a solution, you can try to reduce the playback speed of the bag file, refer to this [blog](https://blog.csdn.net/learning_tortosie/article/details/116051761?spm=1001.2014.3001.5501) for details.

非常感谢，确实降低数据速度可以正常运行，效果不错，可能是由于我计算机cpu的问题不能按照数据包实际的时间运行

closed 2021-04-25 by dzy130120