TixiaoShan/LVI-SAM #3: How did you get the params lidar_to_cam?

hitbg-yjm, 2021-04-26:
params_camera.yaml:
lidar_to_cam_tx: 0.05
lidar_to_cam_ty: -0.07
lidar_to_cam_tz: -0.07
lidar_to_cam_rx: 0.0
lidar_to_cam_ry: 0.0
lidar_to_cam_rz: -0.04

The radar's coordinate system(z) is up, and the camera's coordinate system(z) is forward.
But in the yaml, rpy is 0 0 0.04,
It's almost the same coordinate system.

closed 2021-05-06 by hitbg-yjm

Woodpecker0, 2021-05-09:
@hitbg-yjm I have the same question, have you figure it out?

hitbg-yjm, 2021-05-12:
> @hitbg-yjm I have the same question, have you figure it out?

I don't think it's right, because the parameters are used in the point cloud conversion. Maybe the camera coordinate system is different.

Woodpecker0, 2021-05-17:
> > @hitbg-yjm I have the same question, have you figure it out?
> 
> I don't think it's right, because the parameters are used in the point cloud conversion. Maybe the camera coordinate system is different.

yes, I read the code, and it seems that this transformation is only used as an offset (transform point to the center of imu)