TixiaoShan/LVI-SAM #6: What's PCL version in this repo? Error: [lvi_sam_mapOptmization-7] process has died [pid 72715, exit code -11

xhding1997, 2021-04-27:
I have already built this repo on Ubuntu 20.04 with PCL 1.10 and C++14.
But when I start `rosbag paly garden.bag` , I got these in terminal: 
`
[ INFO] [1619515515.441234729]: ----> Visual Odometry Estimator Started.
[ INFO] [1619515515.465279140]: ----> Lidar IMU Preintegration Started.
[ INFO] [1619515515.475755274]: ----> Lidar Cloud Deskew Started.
[ INFO] [1619515515.478325008]: ----> Lidar Feature Extraction Started.
[ INFO] [1619515515.482130955]: ----> Visual Loop Detection Started.
[ INFO] [1619515515.489539180]: ----> Visual Feature Tracker Started.
[ INFO] [1619515515.503315593]: ----> Lidar Map Optimization Started.
[lvi_sam_mapOptmization-7] process has died [pid 72715, exit code -11, cmd /home/bie/ros_ws/Lvi_Sam/devel/lib/lvi_sam/lvi_sam_mapOptmization __name:=lvi_sam_mapOptmization __log:=/home/bie/.ros/log/79586428-a73a-11eb-9e5c-244bfecec228/lvi_sam_mapOptmization-7.log].
log file: /home/bie/.ros/log/79586428-a73a-11eb-9e5c-244bfecec228/lvi_sam_mapOptmization-7*.log
[lvi_sam_mapOptmization-7] restarting process
process[lvi_sam_mapOptmization-7]: started with pid [72894]
[ INFO] [1619515533.629490956]: ----> Lidar Map Optimization Started.
`
I think it's because my PCL version is different from your repo.

closed 2021-06-03 by stale[bot]

Blackshaddock, 2021-06-07:
I have the same problem， did you solve this problem？ @xhding1997

xuwuzhou, 2021-07-17:
Probably caused by lib of gtsam,  try
$ sudo cp /usr/local/lib/libmetis.so  /usr/lib