TixiaoShan/LVI-SAM #7: Does LVI-SAM really improve performance?

robosu12, 2021-04-28:
After reading your paper and code, thank you very much for your work. But I have the following questions:
1. The laser odometry only uses the result of the visual odometry as the initial optimization value. This should be loosely coupled.
2. Compared with LIO-SAM, LVI-SAM only uses the result of the visual odometry as the initial value of scanning matching, and does not improve the laser odometry. Can the performance of the LVI-SAM really be significantly improved?
3. The results of laser odometry, visual odometry, and IMU pre-integration are used in the paper to perform factor graph optimization. But only the results of the laser odometry and IMU are used in the code. The laser odometry has higher estimation accuracy than the visual odometry, so adding the result of the visual odometry to the factor graph will reduce the optimization accuracy, right?

jaeseok4104, 2021-04-29:
i agree this commant, if VINS-MONO breaks down, LVI-SAM doesn't produce good results

xuwuzhou, 2021-05-06:
Really? Only the laser odometry and IMU are used in the code?

lewisjiang, 2021-05-06:
As of commit [8b78958](https://github.com/TixiaoShan/LVI-SAM/commit/8b789589cc3d392813cc8226f59d656b443c3608), the data flow is like this (correct me if I'm wrong, which is very likely):
1. The VIO publishes odom as topic `"/vins/odometry/imu_propagate_ros"` in [visualization.cpp#L25](https://github.com/TixiaoShan/LVI-SAM/blob/8b789589cc3d392813cc8226f59d656b443c3608/src/visual_odometry/visual_estimator/utility/visualization.cpp#L25). 
2. And it is subcribed only in [imageProjection.cpp#L93](https://github.com/TixiaoShan/LVI-SAM/blob/8b789589cc3d392813cc8226f59d656b443c3608/src/lidar_odometry/imageProjection.cpp#L93) of the LIO module, which collects the subscribed data in `odometryHandler()`, and it further deskews the laser scans and publishes a self-defined cloud msg called `cloudInfo` embedded with the odom information in the same file.
3. In [mapOptmization.cpp#L833](https://github.com/TixiaoShan/LVI-SAM/blob/8b789589cc3d392813cc8226f59d656b443c3608/src/lidar_odometry/mapOptmization.cpp#L833) the odom in the cloud is read to produce `transFinal`, which is an initial guess of the pose. The pose parameters are copied into `transformTobeMapped[6]` and updated in scan-map optimization in [LMOptimization()](https://github.com/TixiaoShan/LVI-SAM/blob/8b789589cc3d392813cc8226f59d656b443c3608/src/lidar_odometry/mapOptmization.cpp#L1251).
4. The pose parameters `transformTobeMapped[6]` are added into isam2 graph as a "to pose" of a between factor in [addOdomFactor()](https://github.com/TixiaoShan/LVI-SAM/blob/8b789589cc3d392813cc8226f59d656b443c3608/src/lidar_odometry/mapOptmization.cpp#L1374).

So in summary, the function `addOdomFactor()` adds a single odometry factor calculated from l, v and i sensors, instead of 3 independent factors as depicted in the paper.

jaeseok4104, 2021-05-06:
@lewisjiang You're right. Only LIO is being added as a factor. The VIO is only used as an initial guess for the LIO. The VIO replaces only the initial guess of the LIO in the LIO-SAM and does not perform any other role.

SiyuanHuang95, 2021-05-24:
[Replay](https://github.com/TixiaoShan/LVI-SAM/issues/7#issuecomment-828854774)

Based on the paper:

If the VIS reports failure, we then switch to IMU measurements for the initial guess.

The failure of VINS should not lead to the bad result? Since in such case, the LVI-SAM should turn into LIO-SAM?

closed 2021-06-09 by stale[bot]