TixiaoShan/LVI-SAM #14: lvi_sam_mapOptmization error

ppwwwwwpp, 2021-05-11:
[lvi_sam_mapOptmization-6] process has died [pid 22275, exit code -6, cmd /home/reeman/wul/LIO-SAM/lvi-sam/devel/lib/lvi_sam/lvi_sam_mapOptmization __name:=lvi_sam_mapOptmization __log:=/home/reeman/.ros/log/64900b78-b201-11eb-ba29-353bc025c460/lvi_sam_mapOptmization-6.log].
log file: /home/reeman/.ros/log/64900b78-b201-11eb-ba29-353bc025c460/lvi_sam_mapOptmization-6*.log

closed 2021-06-03 by stale[bot]

hannibal051, 2021-08-31:
Show your pcl version

skyrim835, 2022-03-25:
> [lvi_sam_mapOptmization-6] process has died [pid 22275, exit code -6, cmd /home/reeman/wul/LIO-SAM/lvi-sam/devel/lib/lvi_sam/lvi_sam_mapOptmization __name:=lvi_sam_mapOptmization [...]

I ran the code in my laptop,and got the same problem,did you solve it?