TixiaoShan/LVI-SAM #15: Can't reproduce result with handheld.bag

smwgf, 2021-05-13:
Thank you for share awesome work.

I can't reproduce result with shared dataset(https://drive.google.com/drive/folders/1q2NZnsgNmezFemoxhHnrDnp1JV_bqrgV?usp=sharing).

It works fine at the beginning. However,  at several points, displayed the warning messages and the trajectory was drifted.
also, I checked library version. (gtsam-4.0.2, ceres-solver-1.14.0)

Could PC specs have an impact?
I installed Linux on Macbook pro A1707.
Processo : 2.6 GHz Intel Core i7 (I7-6700HQ)
RAM: 16GB
GPU: Radeon Pro 450

Warning message:
Large bias, reset IMU-preintegration!
Large velocity, reset IMU-preintegration!

JinMing-Gu, 2021-05-17:
Maybe the speed of data release is too fast.

vk-st, 2021-05-28:
I have a similar problem but different error. In my case it also works fine at the beginning but at some point I receive the following warnings
[ WARN] [1622210777.588924217]: image discontinue! reset the feature tracker!
[ WARN] [1622210777.610756949]: restart the estimator!
[ WARN] [1622210778.491935815]: throw img, only should happen at the beginning
[lvi_sam_visual_loop-9] process has died [pid 3517, exit code -9, cmd /home/skiparissi/catkin_wsLVI/devel/lib/lvi_sam/lvi_sam_visual_loop __name:=lvi_sam_visual_loop __log:=/home/skiparissi/.ros/log/137f13e2-bfbb-11eb-8763-645a04c36a0e/lvi_sam_visual_loop-9.log].
log file: /home/skiparissi/.ros/log/137f13e2-bfbb-11eb-8763-645a04c36a0e/lvi_sam_visual_loop-9*.log

Any ideas?

smwgf, 2021-05-31:
It was solved by changing the rosbag option as shown below.
<rosbag play -r 0.5>

closed 2021-06-20 by stale[bot]