TixiaoShan/LVI-SAM #19: question about  "q_lidar_to_cam q_lidar_to_cam_eigen"

HeXu1, 2021-06-07:
Hi,thanks for your great works. 
the code test well with your data,but it seems the vins program did not work well with my data. I’m confused about what the “question about  q_lidar_to_cam q_lidar_to_cam_eigen” mean in the code. hope your reply ,thanks agian.
@TixiaoShan

HeXu1, 2021-06-07:
[image]
[image]

HeXu1, 2021-06-07:
my imu align lidar well,the first show lvi-sam works with vins but vins failed,the second show lvi-sam works with lio-sam only it works well.

closed 2021-06-26 by stale[bot]

epicjung, 2021-09-07:
> Hi,thanks for your great works.
> the code test well with your data,but it seems the vins program did not work well with my data. I’m confused about what the “question about q_lidar_to_cam q_lidar_to_cam_eigen” mean in the code. hope your reply ,thanks agian.
> @TixiaoShan

Check this repository (https://github.com/epicjung/LVI_SAM_fixed). I have changed the code for it to work for general cases. Please let me know if the modified version does not work for you.

HeXu1, 2021-09-16:
@epicjung thanks a lot. I will test it a few days later.