TixiaoShan/LVI-SAM #21: About external parameters

winnnda, 2021-07-22:
Hello here, the  data is from KITTI dataset, I gave sensor information and internal and external parameters in yaml as required, but every time I open lidar for depth info in params_camera.yaml, the vio part will crash with the following error. Is this a problem with external parameter settings?
[image]

Liuhao91, 2021-07-26:
I have same problem,but i also do not how to solve?Did you solve this problem？

winnnda, 2021-07-26:
> I have same problem,but i also do not how to solve?Did you solve this problem？

Not yet, according to the lidar_to_cam parameters given in params_camera.yaml, the  lidar's and camera's coordinate seem to have similar axial directions, but I can not derive this parameter from the camera--> imu and imu-->lidar parameters provided by the sample configuration.

hellopopyee, 2021-07-27:
I also encountered this problem. I don't think it is an external reference problem, because the available depth has not been extracted from the image display of lidar in rviz. Is it convenient for private communication

Liuhao91, 2021-07-28:
Can we contact by email？843384412@qq.com

winnnda, 2021-07-28:
> I also encountered this problem. I don't think it is an external reference problem, because the available depth has not been extracted from the image display of lidar in rviz. Is it convenient for private communication

Sure, and my email address is 3286336916@qq.com.

closed 2021-08-22 by stale[bot]

epicjung, 2021-09-07:
I have the same problem and modify the code so that extrinsic parameters are correctly used for general cases. Check the repository (https://github.com/epicjung/LVI_SAM_fixed) and see if you can fix the issue.

sw-Tom, 2022-11-13:
Maybe it is helpful
https://zhuanlan.zhihu.com/p/440288323

hamzaceme, 2026-09-17:
anyone who solved this issue for KITTI dataset I am using the lvi-sam-easy used repo but still the error comes. @winnnda were you able to solve this problem if yes can you please guide on how you solved. Any repo which works or what have you done.