TixiaoShan/LVI-SAM #27: Is deskewing correct?

juliangaal, 2021-10-05:
```
  void findPosition(double relTime, float *posXCur, float *posYCur, float *posZCur)
  {
      *posXCur = 0; *posYCur = 0; *posZCur = 0;

      // if (cloudInfo.odomAvailable == false || odomDeskewFlag == false)
      //     return;

      // float ratio = relTime / (timeScanNext - timeScanCur);

      // *posXCur = ratio * odomIncreX;
      // *posYCur = ratio * odomIncreY;
      // *posZCur = ratio * odomIncreZ;
  }
```

As far as I understand, this will result in the pcl being deskewed in rotation only, is this correct?

juliangaal, 2021-10-10:
Doesn't that degrade the accuracy of scan matching and map quality?

tigerschweiger, 2021-10-10:
i feared so, i thought it was super proper for the low-velocity, but in high-way this problem must be taken into account.

closed 2022-04-16 by stale[bot]