TixiaoShan/LVI-SAM #29: large velocity or bias, reset IMU-preintegration!

SnowCarter, 2021-10-13:
There will be a significant drop in the z-axis after around 2mins running. After then, LVI-SAM will become totally unstable. 
Is there anybody who comes across this issue?
[image]
[image]
[image]

SnowCarter, 2021-10-13:
Sorry guys, not the problem of LVI-SAM. It was caused by RVIZ because RVIZ took up too much computing resources.

closed 2021-10-13 by SnowCarter

zilong-ding, 2022-04-12:
我是在楼道内测试，也遇到了这个问题，我是给予了IMU更多信任

Lyk77, 2022-12-13:
@zilong-ding How to do it？I also encountered this problem。

SnowCarter, 2022-12-14:
For me, I just turn off some display options in RVIZ: like turning off all of pointcloud2 msg. However, it can also be because of the IMU noise as the @zilong-ding mentioned. You can redo the IMU calibration and maybe increase the noise value by times 10.

jeychandar, 2022-12-27:
> Sorry guys, not the problem of LVI-SAM. It was caused by RVIZ because RVIZ took up too much computing resources.

Can I know where this error came from. I tried alot but could not find. Having the same issue but I am working with li_slam_ros2 repo. Could you help me out. Thanks in advance.

[image]