TixiaoShan/LVI-SAM #32: process has died

Stephen1e, 2021-11-09:
Hello, when I run LVI Sam on the Jetson NX development board，catkin_make has been completed，but when I tried roslaunch, I encountered the following node death problem

[ INFO] [1636441314.154527795]: ----> Visual Odometry Estimator Started.
[ INFO] [1636441314.487589783]: ----> Visual Loop Detection Started.
[ INFO] [1636441314.585487422]: ----> Visual Feature Tracker Started.
[lvi_sam_visual_odometry-9] process has died [pid 13242, exit code -11, cmd /home/nvidia/catkin_sam/devel/lib/lvi_sam/lvi_sam_visual_odometry __name:=lvi_sam_visual_odometry __log:=/home/nvidia/.ros/log/e982fd3c-412a-11ec-ad02-48b02d3da899/lvi_sam_visual_odometry-9.log].
log file: /home/nvidia/.ros/log/e982fd3c-412a-11ec-ad02-48b02d3da899/lvi_sam_visual_odometry-9*.log
[lvi_sam_visual_odometry-9] restarting process
process[lvi_sam_visual_odometry-9]: started with pid [13528]
[ INFO] [1636441314.871099244]: ----> Visual Odometry Estimator Started.
[ INFO] [1636441314.993614341]: ----> Lidar IMU Preintegration Started.
[lvi_sam_visual_loop-10] process has died [pid 13249, exit code -11, cmd /home/nvidia/catkin_sam/devel/lib/lvi_sam/lvi_sam_visual_loop __name:=lvi_sam_visual_loop __log:=/home/nvidia/.ros/log/e982fd3c-412a-11ec-ad02-48b02d3da899/lvi_sam_visual_loop-10.log].
log file: /home/nvidia/.ros/log/e982fd3c-412a-11ec-ad02-48b02d3da899/lvi_sam_visual_loop-10*.log
[lvi_sam_visual_loop-10] restarting process
[ INFO] [1636441315.091523468]: ----> Lidar Cloud Deskew Started.
process[lvi_sam_visual_loop-10]: started with pid [13703]
[ INFO] [1636441315.169702089]: ----> Lidar Feature Extraction Started.
[lvi_sam_visual_feature-8] process has died [pid 13231, exit code -11, cmd /home/nvidia/catkin_sam/devel/lib/lvi_sam/lvi_sam_visual_feature __name:=lvi_sam_visual_feature __log:=/home/nvidia/.ros/log/e982fd3c-412a-11ec-ad02-48b02d3da899/lvi_sam_visual_feature-8.log].
log file: /home/nvidia/.ros/log/e982fd3c-412a-11ec-ad02-48b02d3da899/lvi_sam_visual_feature-8*.log

My environment is:
Ubuntu18.04
ROS melodic
gtsam4.0.2
ceres1.14.0
opencv4.1.1
pcl1.8

How can I solve the problem？

ZhongKang97, 2021-11-27:
[image]
same issue here

ZhongKang97, 2021-11-27:
the lidar map optimization can not run successfully

ZhongKang97, 2021-11-27:
Oh, I forgot to use -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF when compiling the gtsam lib and after recompile,it works 
[image]

BJM666, 2022-03-14:
Have you solved this problem? I met it ,too. @ZhongKang97

ZhongKang97, 2022-03-14:
> Have you solved this problem? I met it ,too. @ZhongKang97

Yeah, I re-installed the gtsam lib with  additional command -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF. After this, it works.

hyhyhy666, 2022-03-14:
我也遇到了 。。。 就突然 运行不了了

closed 2022-04-17 by stale[bot]