TixiaoShan/LVI-SAM #35: malloc(): memory corruption

DavidNY123, 2022-02-18:
[lvi_sam_mapOptmization-6] process has died [pid 13598, exit code -6, cmd /home/mwy/lvisam/devel/lib/lvi_sam/lvi_sam_mapOptmization __name:=lvi_sam_mapOptmization __log:=/home/mwy/.ros/log/f460d3d2-9099-11ec-a978-9061ae86e6b5/lvi_sam_mapOptmization-6.log].
log file: /home/mwy/.ros/log/f460d3d2-9099-11ec-a978-9061ae86e6b5/lvi_sam_mapOptmization-6*.log

How to solve this problem? I install the gtsam following your introduction. 
I am sure I have used  cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..  
And my pcl version is 1.8.

closed 2022-04-17 by stale[bot]

1528993321, 2022-11-16:
[lvi_sam_mapOptmization-7] restarting process
process[lvi_sam_mapOptmization-7]: started with pid [15361]
malloc(): memory corruption
[lvi_sam_mapOptmization-7] process has died [pid 15361, exit code -6, cmd /home/rjz/lvi_sam_ws/devel/lib/lvi_sam/lvi_sam_mapOptmization __name:=lvi_sam_mapOptmization __log:=/home/rjz/.ros/log/ddcd7812-6593-11ed-80f8-54e1adceded4/lvi_sam_mapOptmization-7.log].
log file: /home/rjz/.ros/log/ddcd7812-6593-11ed-80f8-54e1adceded4/lvi_sam_mapOptmization-7*.log
##########################################################
I get the same question .

1528993321, 2022-11-18:
1. /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h  
::flann::SearchParams *param_k_;    ->  ::flann::SearchParams param_k_;
::flann::SearchParams *param_radius_;   -> ::flann::SearchParams param_radius_;
[https://blog.csdn.net/weixin_41698305/article/details/120651431](url)
2.solve pcl and opencv flann conflict
/usr/local/opencv/include/opencv2/flann.hpp:   
 #include "opencv2/flann/miniflann.hpp" -> `#include "opencv2/flann2/miniflann.hpp"`
/usr/local/opencv/include/opencv2/flann.hpp:
`#include "opencv2/flann/flann_base.hpp"`->#include "opencv2/flann2/flann_base.hpp"
/usr/local/opencv/include/opencv2/features2d.hpp:
`#include `"opencv2/flann/miniflann.hpp"`-> `#include "opencv2/flann2/miniflann.hpp"`
/usr/local/opencv/include/opencv2/flann/miniflann.hpp:
`#include `"opencv2/flann/defines.h"`->  `#include "opencv2/flann2/defines.h"`
become the dir name flann to flann2
`sudo mv /usr/local/opencv/include/opencv2/flann /usr/local/opencv/include/opencv2/flann2`
[https://blog.csdn.net/u010677365/article/details/79894433](url)

DavidNY123, 2022-11-18:
> 

Thanks for your reply, I will have a try.