TixiaoShan/LVI-SAM #41: Some problems about LVI SAM data evaluation

liu19980515, 2022-03-21:
Hello, I have a few questions to ask, thank you:

1. I output GPS data and IMU from the handheld data set_ Correct, it is found that the starting point and the ending point are not at the same point. How to evaluate the accuracy of the method;

2. Slam path is a relative position. How to align it with the ENU of GPS path;

3. The sampling rates of GPS and slam path are different, and the time axis is also misplaced. How does RMSE w.r.t GPS evaluate it.

qinsimon, 2022-03-31:
Same problem

liu19980515, 2022-04-07:
> > > Same problem
> > 
> > 
> > Can I add your contact information，thank you
> 
> 或许我们可以直接在这儿讨论哈哈哈哈哈

哈哈哈，我把LVI-sam跑出来的路径取出来，还有GPS路径取出来，两个图形差了大概45°，他是怎么对齐的

icameling, 2022-04-08:
> > > > Same problem
> > > 
> > > 
> > > Can I add your contact information，thank you
> > 
> > 
> > 或许我们可以直接在这儿讨论哈哈哈哈哈
> 
> 哈哈哈，我把LVI-sam跑出来的路径取出来，还有GPS路径取出来，两个图形差了大概45°，他是怎么对齐的

可以用[evo](https://github.com/MichaelGrupp/evo)工具来对齐轨迹。

liu19980515, 2022-04-08:
> > > > > Same problem
> > > > 
> > > > 
> > > > Can I add your contact information，thank you
> > > 
> > > 
> > > 或许我们可以直接在这儿讨论哈哈哈哈哈
> > 
> > 
> > 哈哈哈，我把LVI-sam跑出来的路径取出来，还有GPS路径取出来，两个图形差了大概45°，他是怎么对齐的
> 
> 可以用[evo](https://github.com/MichaelGrupp/evo)工具来对齐轨迹。

额，这个SLAM没有指北也能和ENU对齐么

icameling, 2022-04-08:
> > > > > > Same problem
> > > > > 
> > > > > 
> > > > > Can I add your contact information，thank you
> > > > 
> > > > 
> > > > 或许我们可以直接在这儿讨论哈哈哈哈哈
> > > 
> > > 
> > > 哈哈哈，我把LVI-sam跑出来的路径取出来，还有GPS路径取出来，两个图形差了大概45°，他是怎么对齐的
> > 
> > 
> > 可以用[evo](https://github.com/MichaelGrupp/evo)工具来对齐轨迹。
> 
> 额，这个SLAM没有指北也能和ENU对齐么

可以的，你可以试试。而且对齐的时候，可以再尝试下添加 --t_offset 0.5 的参数，结果可能会好一些。

liu19980515, 2022-04-11:
> > > > > > > Same problem
> > > > > > 
> > > > > > 
> > > > > > Can I add your contact information，thank you
> > > > > 
> > > > > 
> > > > > 或许我们可以直接在这儿讨论哈哈哈哈哈
> > > > 
> > > > 
> > > > 哈哈哈，我把LVI-sam跑出来的路径取出来，还有GPS路径取出来，两个图形差了大概45°，他是怎么对齐的
> > > 
> > > 
> > > 可以用[evo](https://github.com/MichaelGrupp/evo)工具来对齐轨迹。
> > 
> > 
> > 额，这个SLAM没有指北也能和ENU对齐么
> 
> 我先从bag中取出gps数据
> 
> ```
> rostopic echo /gps/fix -pb handheld.bag > handheld.csv
> ```
> 
> 再用 python 的 utm 工具转换坐标系后保存
> 
> ```
> import numpy as np
> import pandas as pd
> import utm
> data = pd.read_csv('handheld.csv',',')
> 
> lats = np.array(data['field.latitude'])
> lons = np.array(data['field.longitude'])
> easting, northing = utm.from_latlon(lats, lons)[0:2]
> data['easting'] = easting
> data['northing'] = northing 
>     
> data['easting'] = data['easting'] - data['easting'][0] 
> data['northing'] = data['northing'] - data['northing'][0] 
> data['height'] = data['field.altitude'] - data['field.altitude'][0]
> 
> data_out = pd.DataFrame(data, columns=['field.header.stamp', 'easting', 'northing', 'height'])
> data_out['qx'] = 0
> data_out['qy'] = 0
> data_out['qz'] = 0
> data_out['qw'] = 1
> data_out.to_csv('handheld_edited.csv', ' ', header = False, index = False)
> ```
> 
> 之后就可以用 evo 对齐。而且对齐的时候，可以再尝试下添加 --t_offset 0.5 的参数，结果可能会好一些。

那mapping/path您怎么存的呢

icameling, 2022-04-11:
> 那mapping/path您怎么存的呢

我用的其他开源里程计跑的，有直接输出轨迹。

我的困惑是，LVI-SAM 提供的 GPS 真值中有一些数据有跳变，这对评估结果影响巨大，不知道原作者是怎么处理的。

liu19980515, 2022-04-11:
> > 那mapping/path您怎么存的呢
> 
> 我用的其他开源里程计跑的，有直接输出轨迹。
> 
> 我的困惑是，LVI-SAM 提供的 GPS 真值中有一些数据有跳变，这对评估结果影响巨大，不知道原作者是怎么处理的。

GPS高程一直不稳定，应该是忽略高度只做平面误差吧

liu19980515, 2022-04-11:
> > 那mapping/path您怎么存的呢
> 
> 我用的其他开源里程计跑的，有直接输出轨迹。
> 
> 我的困惑是，LVI-SAM 提供的 GPS 真值中有一些数据有跳变，这对评估结果影响巨大，不知道原作者是怎么处理的。

您用的jackal GPS首尾相接么，我没找到jackal的bag包，我用的handheld首位不相接不知道end to end怎么评价

icameling, 2022-04-11:
> 您用的jackal GPS首尾相接么，我没找到jackal的bag包，我用的handheld首位不相接不知道end to end怎么评价

jackal的bag包作者在这儿有提供 https://github.com/TixiaoShan/LVI-SAM#datasets 

 jackal 包首尾点也不相同，也是经过起始点后继续向前运动，可能需要自己找对应位置。

liu19980515, 2022-04-11:
> > 您用的jackal GPS首尾相接么，我没找到jackal的bag包，我用的handheld首位不相接不知道end to end怎么评价
> 
> jackal的bag包作者在这儿有提供 https://github.com/TixiaoShan/LVI-SAM#datasets
> 
> jackal 包首尾点也不相同，也是经过起始点后继续向前运动，可能需要自己找对应位置。

thanks，好家伙没给个开始结束标识么，这自己咋找，不能靠猜把<_<

liu19980515, 2022-04-11:
> > 您用的jackal GPS首尾相接么，我没找到jackal的bag包，我用的handheld首位不相接不知道end to end怎么评价
> 
> jackal的bag包作者在这儿有提供 https://github.com/TixiaoShan/LVI-SAM#datasets
> 
> jackal 包首尾点也不相同，也是经过起始点后继续向前运动，可能需要自己找对应位置。

GPS xy轴怎么也有1m误差吧

closed 2022-04-28 by stale[bot]