TixiaoShan/LVI-SAM #49: Which is the local position output topic?

Cristian-wp, 2022-06-27:
Sorry for the dummy question, but I can not understan which is the position topic.
I have understand that **/lvi_sam/lidar/mapping/odometry** is related to the fusion LIDAR+IMU (LIO-SAM output) and **/lvi_sam/vins/odometry/odometry** is related to CAMERA+IMU (VINS-MONO output).
But which is the one that perform the fusion of both algorithms?

wpwebrobotics, 2022-06-30:
Based on code and this discussion: https://github.com/TixiaoShan/LVI-SAM/issues/7,
I think the topic to look at is /lidar/mapping/odometry published by mapOptmization.cpp as VIO is used only for the initial guess of LIO sub-system.
Can anyone confirm this?

closed 2023-06-18 by stale[bot]

Wangben1019, 2024-03-24:
Have you obtained the answer to this question？Is the final topic /lvi_sam/lidar/mapping/odometry ?