TixiaoShan/LVI-SAM #57: difference between code and article

wjf1997, 2022-09-02:
The sentence "The constraints from visual odometry, lidar odometry, IMU preintegration, and loop closure are optimized jointly in the factor graph" is mentioned in related articles.
but it seems that visual constrains and inertial constrains are not contained in SAM optimization of mapOptimization.cpp file.

closed 2023-06-18 by stale[bot]

wjf1997, 2023-06-29:
cannot agree with you any more

---- Replied Message ----
| From | GAO ***@***.***> |
| Date | 06/29/2023 18:20 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [TixiaoShan/LVI-SAM] difference between code and article (Issue #57) |

Interesting question! I guess it is hard to run in real-time if the visual feature residuals also are included in the factor graph. It is possible to consider the output of the VIS system as one factor, then it is jointly optimized with point-to-line/plane in lidar, IMU preintegration, and loop closure.

shamengjun, 2023-11-08:
Actually, the author in code had commented it as picture :
[image]
if the lidar odometry is degenerate, then consider add visual odometry constraint, but the benefits are not obvious.
I think vins odometry and lidar odometry output frequency is strictly as the same, so the pose alignment error can cause extra error, so the benefits are not obvious?