TixiaoShan/LVI-SAM #64: Clarity on Licensing

mcamurri, 2023-05-09:
Dear @TixiaoShan thanks for the great code and awesome results. LVI-SAM as repository is presented as BSD3 code, but down below it says the visual odometry component is adapted from VINS-Mono, which is GPL3. 

To avoid legal issues any sources derivative from VINS-mono should therefore be redistributed as GPL3.

yinhaosen, 2023-08-10:
The author may have adapted VINS-Mono to a distinguished extent from the original one.

yhu-bdai, 2025-06-19:
The visual odometry part does significantly similar to the VINS-Mono original implementation...

mcamurri, 2025-06-19:
Any derivative implementation of a GPL source code has to be GPL, irrespective of how big its difference from the original is. 
So even more so if the difference is small!